udoklein/dcf77 #13: MSF60 development code takes very long time to lock to signal

aido, 2015-09-15:
Hi Udo/Gavin,

I have been testing the MSF60 code in the development branch and find that it can take several days to lock on to an MSF signal. I initially tested the signal using DCF Scope and found that the MSF signal I receive is almost perfect. However, when testing MSF60 using the Simple Clock code (with `dcf77_inverted_samples=0`) I find that the clock can take several days to lock onto the MSF signal and then decodes the wrong time.

Regards,
Aido

udoklein, 2015-09-17:
Hi Aido,

I have sent a personal mail to Gavin and asked if he knows why this happens. So far I have no good means to test this. RIght now I am creating massive scaffolding to nail down any errors. So far I am only 30% done. So I will take some more time. 

But thanks a lot for reporting this.

Regards, Udo

udoklein, 2015-09-17:
Hi Aido,

Gavin suspects that this is an issue with your setup. Maybe a resonator 
instead of a crystal or a crystal out of spec? Can we have a trace with 
"swiss army debug helper"? Especially in the "scope" mode?

Best regards, Udo

udoklein, 2015-09-17:
I just ran the code again... using development branch and SimpleClock which is slightly modified to use Pin 8and to enable the receiver.  I attach the source.
The only thing to bear in mind is that you must #define MSF60  1
The receiver takes around 10 mins to get a lock and will need a quite spot well away from a computer monitor and with the antenna aligned at 90 degrees to the transmitter.
Hope that helps
Please cc: Aido
RegardsGavin

      From: Udo Klein <mail@udo-klein.net>
 To: udoklein/dcf77 <reply@reply.github.com> 
Cc: Gavin Andrews <GavinAndrews@yahoo.com> 
 Sent: Thursday, 17 September 2015, 18:50
 Subject: Re: [dcf77] MSF60 development code takes very long time to lock to signal (#13)

Hi Aido,

Gavin suspects that this is an issue with your setup. Maybe a resonator 
instead of a crystal or a crystal out of spec? Can we have a trace with 
"swiss army debug helper"? Especially in the "scope" mode?

Best regards, Udo

udoklein, 2015-09-19:
Hi Aido,
I do not have a MSF receiver. To analyze the issue I would need to see the signal. Can you please send us logs created by the Swiss Army Debug Helper? Preferably at least 1 hour in mode Ds followed by at least 15 minutes in mode Dd?
Regards, Udo

GavinAndrews, 2015-09-19:
Aido

I have it working with a Chinese Uno clone but only after changing the resonator for a xtal.  With the resonator, which I first thought was a crystal due to its shiny nature, I found the clock was 2000 parts per million out and this was enough to cause the lock to fail.  Please can you post an image of your board, or if you have a scope we run a test to check the frequency.

Also Make sure you #define MSF60 1.

I have the following set-up for SimpleClock

#if defined(__AVR__)
const uint8_t dcf77_analog_sample_pin = 5;
const uint8_t dcf77_sample_pin = 8;       // A5 == d19
const uint8_t dcf77_inverted_samples = 0;
const uint8_t dcf77_analog_samples = 0;
const uint8_t dcf77_pull_up = 1;

const uint8_t dcf77_monitor_led = 13;  // A4 == d18

uint8_t ledpin(const uint8_t led) {
    return led;
}

I use the module from PV Electronics, powered from 5v with input to pin 8 and using pin 7 as output to turn the module on an off.

and I have...
    pinMode(13, OUTPUT);   // On-board LED
    pinMode(8, INPUT);     // Data In
    pinMode(7, OUTPUT);     // Power Control to MSF Receiver
    digitalWrite(7, LOW);   // Enable MSF Receiver    

With a clean signal (well away from a computer monitor or switched mode supply it takes around 10 mins to get a lock.

[image]

It would be really good to pursue this and get the MSF60 part robust.

Thanks

aido, 2015-09-19:
Hi Gavin,

I have tested with an Arduino micro board. According to the tech specs this has a 16 MHz crystal oscillator.
https://www.arduino.cc/en/Main/ArduinoBoardMicro#techspecs

Board was purchased from a reputable source (Adafruit) so unlikely to be a cheap knockoff with a ceramic resonator.

I will send log output of Swiss Army Debug Helper soon.

Regards,
    Aido

GavinAndrews, 2015-09-19:
The signal looks good, now slip, so crystal seems fine.

Can you make me a capture using the below code...  You will need to change the pin assignments to match your set-up, the input pin doesn't need to be ICP.

```
int ledPin = 13;      // select the pin for the LED

volatile int kick = 0;

#define icpPin 8        // ICP input pin on arduino

void setup() {

  pinMode(ledPin, OUTPUT);

  pinMode(icpPin, INPUT);
  pinMode(7, OUTPUT);     // Power Control to MSF Receiver
  digitalWrite(7, LOW);   // Enable MSF Receiver

  Serial.begin(19200);

  TCCR1A = 0 ;           // normal mode
  TCCR1B = _BV(CS10) | _BV(WGM12);    // CTC MODE  CS12=0,CS11=0,CS10=1 => clk/1
  TCCR1C = 0;            // normal mode

  OCR1A = 15999;

  TCNT1 = 0;                     // Start from Zero
  TIMSK1 =  _BV(OCIE1A);         // Compare Enable
}

int   activeBucketSet = 0;
byte* lastBucketSet = 0;

volatile byte bucketSet = 0;

volatile byte index = 0;
volatile byte bucket = 0;

byte buckets_a[100];
byte buckets_b[100];
byte* activeBucket = &buckets_a[0];

ISR(TIMER1_COMPA_vect) {

  byte val = digitalRead(icpPin);
  val = (val == 0) ? 1 : 16;

  if (index == 0) {
    *activeBucket = 0;
  }

  *activeBucket += val;

  index++;
  if (index >= 10) {
    activeBucket++;
    bucket++;
    index = 0;
  }
  
  if (bucket>99) {
    
    lastBucketSet = (activeBucketSet == 0) ? &buckets_a[0] : &buckets_b[0];
    kick = 1;    
    digitalWrite(13, !digitalRead(13));   // toggle state of Pin 13  // read the value from the sensor:

    activeBucketSet = 1 - activeBucketSet;
    
    index = 0;
    bucket = 0;
    activeBucket = (activeBucketSet == 0) ? &buckets_a[0] : &buckets_b[0];
  }
}

int second = 0;

void loop() {

  if (kick == 1) {
    if (second < 10) Serial.print("0");
    Serial.print(second);
    Serial.print(" ");

    for (int i = 0; i < 100; i++)
    {
      byte v = lastBucketSet[i];
      byte lo = (v & 0xf);
      byte hi = (v >> 4);

      Serial.print(char(65 + hi));
      Serial.print(char(65 + lo));
    }

    Serial.println("");

    second++;
    if (second >= 60) second = 0;

    kick = 0;
  }
}

```

Try and capture an hours worth.

Thanks
Gavin

aido, 2015-09-20:
Hi Gavin,

Running following code:
```C
int ledPin = 13;      // select the pin for the LED

volatile int kick = 0;

#define icpPin A5        // ICP input pin on arduino

void setup() {

  pinMode(ledPin, OUTPUT);

  pinMode(icpPin, INPUT);
//  pinMode(7, OUTPUT);     // Power Control to MSF Receiver
//  digitalWrite(7, LOW);   // Enable MSF Receiver
.
.
.
```
I changed the input pin to A5. I also have the PON pin of the MSF module permanently grounded so no need for power control using Arduino pin 7.

Will upload output shortly.

Regards,
Aido

GavinAndrews, 2015-09-20:
I took your log, converted it to mS form and added it to the standalone test harness... and got...

```
C:\Users\Gavin\.clion10\system\cmake\generated\c177209b\c177209b\Debug2\dcf77.exe
OutTime: 20/09/2015 01:57:00 Summer
OutTime: 20/09/2015 01:58:00 Summer
OutTime: 20/09/2015 01:59:00 Summer
OutTime: 20/09/2015 02:00:00 Summer
OutTime: 20/09/2015 02:01:00 Summer
OutTime: 20/09/2015 02:02:00 Summer
OutTime: 20/09/2015 02:03:00 Summer
OutTime: 20/09/2015 02:04:00 Summer
OutTime: 20/09/2015 02:05:00 Summer
OutTime: 20/09/2015 02:06:00 Summer
OutTime: 20/09/2015 02:07:00 Summer
OutTime: 20/09/2015 02:08:00 Summer
OutTime: 20/09/2015 02:09:00 Summer
OutTime: 20/09/2015 02:10:00 Summer
OutTime: 20/09/2015 02:11:00 Summer
OutTime: 20/09/2015 02:12:00 Summer
OutTime: 20/09/2015 02:13:00 Summer
OutTime: 20/09/2015 02:14:00 Summer
OutTime: 20/09/2015 02:15:00 Summer
OutTime: 20/09/2015 02:16:00 Summer
OutTime: 20/09/2015 02:17:00 Summer
OutTime: 20/09/2015 02:18:00 Summer
OutTime: 20/09/2015 02:19:00 Summer
OutTime: 20/09/2015 02:20:00 Summer
OutTime: 20/09/2015 02:21:00 Summer
OutTime: 20/09/2015 02:22:00 Summer
OutTime: 20/09/2015 02:23:00 Summer
OutTime: 20/09/2015 02:24:00 Summer
OutTime: 20/09/2015 02:25:00 Summer
OutTime: 20/09/2015 02:26:00 Summer
OutTime: 20/09/2015 02:27:00 Summer
OutTime: 20/09/2015 02:28:00 Summer
OutTime: 20/09/2015 02:29:00 Summer
OutTime: 20/09/2015 02:30:00 Summer
OutTime: 20/09/2015 02:31:00 Summer
OutTime: 20/09/2015 02:32:00 Summer
OutTime: 20/09/2015 02:33:00 Summer
OutTime: 20/09/2015 02:34:00 Summer
OutTime: 20/09/2015 02:35:00 Summer
OutTime: 20/09/2015 02:36:00 Summer
OutTime: 20/09/2015 02:37:00 Summer
OutTime: 20/09/2015 02:38:00 Summer
OutTime: 20/09/2015 02:39:00 Summer
OutTime: 20/09/2015 02:40:00 Summer
OutTime: 20/09/2015 02:41:00 Summer
OutTime: 20/09/2015 02:42:00 Summer
OutTime: 20/09/2015 02:43:00 Summer
OutTime: 20/09/2015 02:44:00 Summer
OutTime: 20/09/2015 02:45:00 Summer
OutTime: 20/09/2015 02:46:00 Summer
OutTime: 20/09/2015 02:47:00 Summer
OutTime: 20/09/2015 02:48:00 Summer
OutTime: 20/09/2015 02:49:00 Summer
OutTime: 20/09/2015 02:50:00 Summer
OutTime: 20/09/2015 02:51:00 Summer
OutTime: 20/09/2015 02:52:00 Summer
OutTime: 20/09/2015 02:53:00 Summer
OutTime: 20/09/2015 02:54:00 Summer
OutTime: 20/09/2015 02:55:00 Summer
OutTime: 20/09/2015 02:56:00 Summer
OutTime: 20/09/2015 02:57:00 Summer
OutTime: 20/09/2015 02:58:00 Summer
OutTime: 20/09/2015 02:59:00 Summer
OutTime: 20/09/2015 03:00:00 Summer
OutTime: 20/09/2015 03:01:00 Summer

Process finished with exit code 0
```

Very weird, looks like it can decode it correctly.

GavinAndrews, 2015-09-20:
I converted your log to a .csv format suitable for the standalone test harness, commited it and raised a pull request to the main repository.  It decodes OK (the test hardness logs each minute after a sync) so all is good with the decode logic and you can see that your crystal is perfect.

I wonder if it is something to do with the ampling or timers on your platform?

GavinAndrews, 2015-09-20:
Are you sure you have MSF60 defined?  Maybe just put it at the top of dcf77.h for now?

aido, 2015-09-20:
Hi Gavin,

To be absolutely certain that I have MSF60 defined I added following to the Swiss Army Debug Helper code:
```C
.
.
.
    void help() {
        Serial.println();
        #ifdef MSF60
        Serial.println(F("MSF60"));
        #else
        Serial.println(F("DCF77"));
        #endif
        Serial.println(F("====="));
        Serial.println();
        Serial.println(F("use serial interface to alter settings"));
.
.
.
```

Output from pressing 'h' in Swiss Army Debug Helper:
```
.
.
.
Tick: 135
  ??.??.??(?,?)??:??:255 MEZ 0,0 0 p(4804-0:255) s(2-2:0) m(0-0:0) h(0-0:0) wd(0-0:0) D(0-0:0) M(0-0:0) Y(0-0:0) 0,0,0,255
Clock state: useless
Tick: 139

MSF60
=====

use serial interface to alter settings
  L: led output modes
    q: quiet
    f: flash
    t: ticks
    2: 200 ms of the signal
 .
 .
 .
```
I am using an Arduino micro board. On these boards timer2 is used by USB so the development code is currently using timer3 instead:
https://github.com/udoklein/dcf77/blob/development/dcf77.cpp#L2378

 I am now testing using timer1 instead but don't think this will make a difference.

Regards,
Aido

GavinAndrews, 2015-09-20:
The critical part of having MSF60 defined when you compile dcf77.cpp.   Where did you #define it?

Also, can you just say which version of Arduino IDE/gcc you are using?  (In case that is another variable)

Like you, I'm not convinced by the timer differences.... confused!

GavinAndrews, 2015-09-20:
```
            TCCR3B = (0<<WGM33) | (1<<WGM32) | (1<<CS31) | (1<<CS30);
            TCCR3A = (0<<WGM31) | (0<<WGM30);
```

Is that correct?  Gives WGM 0100 ?

aido, 2015-09-20:
Hi Gavin,

Problem solved. I had MSF60 defined at the top of `Swiss_Army_Debug_Helper.ino`. On your advice I moved the definition to the top of `dcf77.h`.

BINGO! MSF signal now syncing in about 20 minutes on an Arduino micro board.

Thanks a lot for the assistance and apologies for my stupidity,

Regards,
Aido

GavinAndrews, 2015-09-20:
No worries.  Feel free to refine the sync mark and convolution binning functions (they are based on your code)

:grinning:

closed 2015-09-20 by aido