udoklein/dcf77 #15: Compiler error in examples The_Clock and Simple_Clock

domhardt, 2015-11-07:
*Simple_Clock*
´´´´
Simple_Clock.ino: In function 'void setup()':
Simple_Clock.ino:59:21: error: 'Clock' is not a namespace-name
Simple_Clock.ino:59:26: error: expected namespace-name before ';' token
Simple_Clock.ino:89:26: error: 'Clock' has not been declared
Simple_Clock.ino:90:18: error: 'Clock' has not been declared
Simple_Clock.ino:90:45: error: 'Clock' has not been declared
Simple_Clock.ino:94:9: error: 'Clock' has not been declared
Simple_Clock.ino:94:23: error: expected ';' before 'now'
Simple_Clock.ino:95:39: error: 'now' was not declared in this scope
Simple_Clock.ino: In function 'void loop()':
Simple_Clock.ino:114:5: error: 'Clock' has not been declared
Simple_Clock.ino:114:19: error: expected ';' before 'now'
Simple_Clock.ino:116:35: error: 'now' was not declared in this scope
Simple_Clock.ino:119:18: error: 'Clock' has not been declared
Simple_Clock.ino:120:18: error: 'Clock' has not been declared
Simple_Clock.ino:121:18: error: 'Clock' has not been declared
Simple_Clock.ino:122:18: error: 'Clock' has not been declared
´´´´

*The_Clock*
´´´´
The_Clock.ino:91:31: error: 'Clock' does not name a type
The_Clock.ino:91:45: error: expected unqualified-id before '&' token
The_Clock.ino:91:45: error: expected ')' before '&' token
The_Clock.ino:91:46: error: expected initializer before 'decoded_time'
The_Clock.ino: In function 'void MaxTicks::setup()':
The_Clock.ino:120:41: error: 'output_handler' was not declared in this scope
The_Clock.ino: At global scope:
The_Clock.ino:125:31: error: 'Clock' does not name a type
The_Clock.ino:125:45: error: expected unqualified-id before '&' token
The_Clock.ino:125:45: error: expected ')' before '&' token
The_Clock.ino:125:46: error: expected initializer before 'decoded_time'
The_Clock.ino: In function 'void Ticks::setup()':
The_Clock.ino:150:41: error: 'output_handler' was not declared in this scope
The_Clock.ino: At global scope:
The_Clock.ino:155:31: error: 'Clock' does not name a type
The_Clock.ino:155:45: error: expected unqualified-id before '&' token
The_Clock.ino:155:45: error: expected ')' before '&' token
The_Clock.ino:155:46: error: expected initializer before 'decoded_time'
The_Clock.ino:385:1: error: expected '}' at end of input
´´´´

domhardt, 2015-11-07:
New try with IDE Arduino 1.6.6 from arduino.cc leads to same problems. DCF77_Scope is working.

```
Arduino: 1.6.6 (Mac OS X), Board: "Arduino/Genuino Uno"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/michaeldomhardt/Documents/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/michaeldomhardt/Documents/Arduino/dcf77-udoklein/examples/The_Clock/The_Clock.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/michaeldomhardt/Documents/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/michaeldomhardt/Documents/Arduino/dcf77-udoklein/examples/The_Clock/The_Clock.ino"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp/sketch/The_Clock.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein" "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp/sketch/The_Clock.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein" "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp/sketch/The_Clock.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein" "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp/sketch/The_Clock.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein" "/Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein/dcf77.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein" "/var/folders/8z/tb7k5_xx3y1b_b99mj7h08dc0000gn/T/build3342978b5fc7ef8a94d607cb5554483f.tmp/sketch/The_Clock.ino.cpp"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_ [... cut ...]
                                             ^
The_Clock:155: error: expected ')' before '&' token
The_Clock:155: error: expected initializer before 'decoded_time'
     void output_handler(const Clock::time_t &decoded_time) {
                                              ^
The_Clock:385: error: expected '}' at end of input
 }
 ^
Bibliothek dcf77-udoklein im Ordner: /Users/michaeldomhardt/Documents/Arduino/libraries/dcf77-udoklein (legacy) wird verwendet
exit status 1
'Clock' does not name a type
```

udoklein, 2015-11-07:
The compiler complains that namespace "Clock" is missing. However in dcf77.h this namespace is declared. So either you modified my code or you are compiling new examples vs. an outdated version of my library.

closed 2015-11-07 by udoklein

nospam2000, 2015-11-07:
Are your sure you have only one copy of dcf77.h in your Arduino library subdirectories?
I'm not sure about the recent changes in Arduino, but in former times the 'main' header file had to be named the same as library folder. Unfortunately there are multiple dcf77 libraries which use a header called dcf77.h. This is why I renamed Udo's dcf77.h to dcf77uk.h in my local installation, so I have
dcf77/dcf77.h and dcf77uk/dcf77uk.h
unfortunately I also had to change all references.

domhardt, 2015-11-07:
Hi again,

the dcf77 library in my installation was indeed outdated. So the example from a new pull won't compile. It's working now. Thanks for the hint and sorry for the noise.

Regards,
Michael

udoklein, 2015-11-07:
Never mind. In case you feel you want to improve your karma: http://blog.blinkenlight.net/good-karma/ ;)