udoklein/dcf77 #16: dcf77 library incompatible with 'Tone' command?

mrohner, 2016-01-28:
Hi, In my dcf77 alarm clock scetch  I use the following code:
 void sound_Alarm() {
     tone(ALARM, 2000);
     delay(50);
     noTone(ALARM);
     if (digitalRead(panic_Pin)) alarm = LOW;
     delay(50);
  }

The tone command though seems incompatible with the dcf77 library. I get this when I compile.Any idea. Thanks for your great work.

C:\Users\MARKUS~1\AppData\Local\Temp\build1c605204d1f6abd64fba3c74d60f8a93.tmp/core\core.a(Tone.cpp.o): In function `__vector_7':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Tone.cpp:539: multiple definition of `__vector_7'

C:\Users\MARKUS~1\AppData\Local\Temp\build1c605204d1f6abd64fba3c74d60f8a93.tmp\libraries\dcf77\dcf77.cpp.o:C:\Users\Markus Rohner\Documents\Arduino\libraries\dcf77/dcf77.cpp:2474: first defined here

collect2.exe: error: ld returned 1 exit status

Using library dcf77 in folder: C:\Users\Markus Rohner\Documents\Arduino\libraries\dcf77 (legacy)
Using library Adafruit_NeoPixel in folder: C:\Users\Markus Rohner\Documents\Arduino\libraries\Adafruit_NeoPixel (legacy)
Using library SimpleTimer in folder: C:\Users\Markus Rohner\Documents\Arduino\libraries\SimpleTimer (legacy)
Using library EEPROM at version 2.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM 
exit status 1
Error compiling.

udoklein, 2016-01-28:
Both the tone library as well as my library try to hook ISR(TIMER2_COMPA_vect). Obviously only one of them can get the vector. Now you might want to ask for switching my library to another timer. However the Atmega 328 aka Arduino has only 3 timers which already are all in use. In my opinion timer 2 is the least hogged timer thus I chose this one for my library.
With the Arduino Due I hook into systicks thus I do not mess with any of the timers. I am sorry but as the Arduino (Uno) timers are so crowded I have no better solution.

closed 2016-01-28 by udoklein

mrohner, 2016-01-29:
Thanks for the explanation. Now I'm using the toneAC library instead which is using timer1 hence eliminating the conflict.