udoklein/dcf77 #17: NANO with DCF77-receiver and too many drift (DCF77_Scope)...?

Jo-Achim, 2016-09-22:
Hi,
I have a little problem with the Libraries v.3.0.0 and before on my Arduino Nanos (IDE 1.6.11). ;-)

E.g. "Simple_Clock" will not synchronize; so I see "Initializing..." and a lot of '.' only. 

Because my Nanos with Conrad-DCF77-receiver are working (https://arduino-hannover.de/2012/06/14/dcf77-empfanger-mit-arduino-betreiben/) with synchronization normally within 2 - 3 minutes), ...

Looking at the output from "DCF77_Scope", I think, I found the reason: the output drifts to the right side within 1.000 seconds round about 340 ms.

        1, +---------+---------+---------+---------+---------+---------+---------+---------+---------+---------
        2, +---------+---------+---------+---------+----9XXXXXXXXXXX---+---------+---------+---------+---------
        3, +---------+---------+---------+---------+----9XXXXXXXXXXXXXXXXXXXXX9--+---------+---------+---------
        4, +---------+---------+---------+---------+----7XXXXXXXXXXXXXXXXXXXXX3--+---------+---------+---------
        5, +---------+---------+---------+---------+----9XXXXXXXXXXXX--+---------+---------+---------+---------

      995, +---------+---------+---------+---------+---------+---------+---------+--------7XXXXXXXXXXXXXXXXXXXX
      996, XX--------+---------+---------+---------+---------+---------+---------+--------3XXXXXXXXXXXX5-------
      997, +---------+---------+---------+---------+---------+---------+---------+--------2XXXXXXXXXXXX6-------
      998, +---------+---------+---------+---------+---------+---------+---------+--------3XXXXXXXXXXXXXXXXXXXX
      999, X8--------+---------+---------+---------+---------+---------+---------+--------2XXXXXXXXXXXXXXXXXXXX
     1000, XX3-------+---------+---------+---------+---------+---------+---------+--------2XXXXXXXXXXX8-------- 

This result I get with my best Nano (with the lowest drift)! (I have 3 Nanos from 2 different manufacturers tested.) 

Is my suggestion correct that the 'no-synchronize'-problem is based on this drift? 

Can you say, how high is, based on your filter-algorithm, the maximum drift per unit of time / or the max. tolerable divergences of the 16-MHz oscillator in Hz or in ppm? 
Are they any variables available to solve the problem in software?

Thank you for help.

Best regards, Joachim.

udoklein, 2016-09-22:
340 ms / 1000 s = 340 ppm. This is an indicator that you are running with a resonator instead of a crystal. The issues is documented here: https://blog.blinkenlight.net/experiments/dcf77/dcf77-library/. Search for "Hardware incompatibilities". Solution: get a board with a crystal or replace the resonator by a crystal + proper load capacitors.

Jo-Achim, 2016-09-23:
Hello Udo,

thank you for information's. 

First I will see to get an original (Gravitech) Nano - hopeful with crystal oscillator and proper load capacitors - for reference. 

Best regards, Joachim.

PS: Great work!

closed 2016-09-23 by Jo-Achim

udoklein, 2016-09-24:
According to the Eagle files the Gravitech has a resonator. You may be 
luckier at *bay:

http://www.ebay.de/itm/New-design-Pro-Mini-atmega328-5V-16M-Replace-ATmega128-Arduino-Compatible-Nano-/172271346955?hash=item281c2c250b:g:0S8AAOSwo6lWL0gM

has a crystal and is very cheap. If you like my project you can donate 
here: https://blog.blinkenlight.net/good-karma/

Best regards, Udo