udoklein/dcf77 #23: Examples superfilter and mb_emulator fail to compile with 1.8.1 and 1.8.2

udoklein, 2017-04-03:
Two examples fail to compile for library version 3.1.4.
This bug was reported by Emmanuel A. Neonakis on my blog: https://blog.blinkenlight.net/2015/10/01/improving-quality/#comment-56039. I will fix it.

aido, 2017-04-03:
Hi,

Does this commit to the development branch help any?
https://github.com/udoklein/dcf77/commit/4aff4fefc1e9f997d458b29c507b47945b7788a3

udoklein, 2017-04-04:
The bugs are now fixed: https://github.com/udoklein/dcf77/releases/tag/v3.1.5

With regard to the Leonardo support: I am still lacking a Leonardo to test it. I think I will put it into a branch "experimental Leonardo support" and see what the feedback is. What do you think?

aido, 2017-04-04:
Hi Udo,

The code in commit https://github.com/udoklein/dcf77/commit/4aff4fefc1e9f997d458b29c507b47945b7788a3 on the development branch is tested and works fine on my micro board which shares same interrupts as Leonardo (both are ATmega32u4 based boards).
Commit  https://github.com/udoklein/dcf77/commit/4aff4fefc1e9f997d458b29c507b47945b7788a3 was part of a larger effort to get MSF60 working on a micro board,

I haven't been following development that closely lately so not sure how much master branch has diverged for development branch. So may not be a case of a simple merge of this commit.

udoklein, 2017-04-04:
Yeah, I know. This was unfortunate timing on my site. While you were developing I did some large scale refactoring and never found the time to merge your stuff :(  Then I had some private issues which required more attention. Thus we now have this split state. I think I can merge in the Leonardo stuff though.

udoklein, 2017-04-06:
I added https://github.com/udoklein/dcf77/tree/leonardo_support
Now I am waiting if someone can confirm that this is OK. It compiles OK but as I said I can not test it.

udoklein, 2017-04-13:
I now take the stance that it works till somebody complains.

closed 2017-04-13 by udoklein