udoklein/dcf77 #29: Winter Time

uzi18, 2017-09-02:
Hi Udo!
Do we need to convert time to wintertime if uses_summertime is false?
Does DCF77 send CET time durring summer and CEST time durring winter?
It is unclear for me.

udoklein, 2017-09-04:
DCF77 distributes the official german time signal. Hence CEST in summer and CET in winter. The flag is only there to be able to convert into CET. You could do this without the flag anyway but for the two hours that appear in CET and CEST.

uzi18, 2017-09-04:
So this is local time for Germany  and there is no need to adjust it
from/to wintertime there for example Berlin?

I was confused because signal also has information about
wintertime/summertime and didn't know if I need to change time internally
 after this flags change.
Also found this uses_summertime variable...

Thanks for answare

<https://github.com/udoklein/dcf77/issues/29#issuecomment-326996657>, or mute
the thread

udoklein, 2017-09-04:
You might want to read this:
https://blog.blinkenlight.net/experiments/dcf77/  or http://www.ptb.de/cms/ptb/fachabteilungen/abt4/fb-44/ag-442/verbreitung-der-gesetzlichen-zeit/dcf77/zeitcode.html or https://en.wikipedia.org/wiki/DCF77 for more details or http://www.ptb.de/cms/fileadmin/internet/fachabteilungen/abteilung_4/4.4_zeit_und_frequenz/pdf/2004_Piester_-_PTB-Mitteilungen_114.pdf

Conclusion: this is *the definition* of the local legal time in Germany.

uzi18, 2017-09-04:
> You might want to read this:
> https://blog.blinkenlight.net/experiments/dcf77/ or
> http://www.ptb.de/cms/ptb/fachabteilungen/abt4/fb-44/ag-
> 442/verbreitung-der-gesetzlichen-zeit/dcf77/zeitcode.html or
> https://en.wikipedia.org/wiki/DCF77 for more details or
> http://www.ptb.de/cms/fileadmin/internet/fachabteilungen/abteilung_4/4.
> 4_zeit_und_frequenz/pdf/2004_Piester_-_PTB-Mitteilungen_114.pdf
>
> Conclusion: this is *the definition* of the local legal time in Germany.

Found answere here on http://www.dcf77logs.de in files Sommerzeit.log and
Winterzeit.log, they are very clearly.
Why didn't get there before, realy don't know.
Thanks.

closed 2017-09-26 by uzi18