udoklein/dcf77 #30: Local_Clock not initialized

nameoftherose, 2017-10-23:
I noticed that the Local_Clock is not initialized, its setup method (dcf77.h line 1323) is never being executed.
Though harmless, is it appropriate?

udoklein, 2017-11-05:
The local clock is automatically initialized. However from a consistency point of view I think the clock controller should call the setup anyway. The setup method was introduced because everything else also has a setup method. This can be useful during testing.

Fixing the issue might make the code slightly larger without adding any benefits. I will test a little bit on the size impact.

nameoftherose, 2017-11-07:
I have done the modification you describe, but now I think it is better to leave it as is. The sketch should call it if necessary.

closed 2018-08-09 by udoklein