udoklein/dcf77 #31: Resetting the whole library

lakeroe, 2018-05-10:
Hello,

I'm running the library on an ARM STM32F103C8T6 and this works very well, but now I've noticed a strange behaviour.

After power-on the first received time if (clock_state >= Clock::free) is always valid and accurate but after some hours (or days) the time starts to drift away and sometimes also the date is wrong allthough the inaccuracy should be below 200 ms according to the comment in dcf77.h. This seems to happen especially in a quite noisy environment.

I assume this problem is related to: https://github.com/udoklein/dcf77/issues/27

Because I use this library to synchronize my ARM built-in realtime-clock I need a reliable time only once in a while. So my idea is to reset the library everytime a valid time is received (clock_state >= Clock::free).
My C++ knowledge is quite limited but I've managed to add a reset-function which calls DCF77_Clock::setup() and DCF77_Local_Clock::setup() and I call this reset-function everytime a valid time is received.

I'm currently running a test for about 24h and this seems to work well.

Do you think this is the right approach for my problem and is the reset-function correctly implemented ?

Thanks and best regards,
lakeroe

udoklein, 2018-05-10:
My suspicion would be that this is caused by an unstable local oscillator. You are right that a similar issue was described https://github.com/udoklein/dcf77/issues/27 in the past. However I was never able to reproduce it at all. Without any means to reproduce it I have no way to locate the issue and fix it.

I suggest that you first try to tell the library that your oscillator might be unstable, e.g. set has_stable_ambient_temperature to false.

https://github.com/udoklein/dcf77/blob/1103b38718e8be451031f6b9986a4291c0a085e0/dcf77.h#L123

If this fixes the problem tell me about it. If this does not fix the problem I have no idea what is wrong. If you really need to enforce a reset of the libary this is in fact done by calling `DCF77_Clock::setup()`. This may work around your issue. So for you this might be good enough. For me this is really anoying because I still have no clue why my library fails for some people.

Most probably I need someone who can reproduce the issue and who lives close enough me such that I could visit and instrument the hardware with suitable measurement equipment.

lakeroe, 2018-05-10:
Thanks for your reply.
I will do further tests and report back in a few days ...

lakeroe, 2018-05-14:
I've done some more testing on two different hardwares:

Board 1: Arduino clone (but definitly quartz crystal based)
Board 2: STM32F103C8 blue pill board (http://wiki.stm32duino.com/index.php?title=Blue_Pill)

For all tests I used two different "Reichelt modules" and following time as a reference: https://www.uhrzeit.org/atomuhr.php

1st test: has_stable_ambient_temperature = false (as you suggested)
  I powered-up the boards and waited for the best possible signal quality (clock_state==synced). Then I wrapped the antennas in aluminium foil to disable dcf77-reception. I verified this by the non-blinking dcf77_monitor_led.
  Result after about 24h
  Board 1: clock deviation ~  0 seconds, correct date, clock_state=2 (free)
  Board 2: clock deviation ~ +4 seconds, correct date, clock_state=2 (free)

2nd test: has_stable_ambient_temperature = true
  Same procedure as before but changed software.
  Result after about 24h
  Board 1: clock deviation ~  0 seconds, correct date, clock_state=2 (free)
  Board 2: clock deviation ~ +4 seconds, correct date, clock_state=2 (free)

3rd test: has_stable_ambient_temperature = true
  I powered-up the boards and waited for the best possible signal quality (clock_state==synced). Then I put the antennas in an upright position to simulate worse dcf77-reception.
  Result after about 22h
  Board 1: clock deviation ~ +94 minutes, correct date,           clock_state=3 (unlocked)
  Board 2: clock deviation ~  -4 seconds, wrong date (14.7.2053), clock_state=4 (locked)
           exactly 62304s after power-on the date changes from 14.5.2018 to 14.7.2053

My modest opinion:
The first two tests show similar behaviour. Allthough the clock deviation on the second board is quite large it should be okay.
According to your github page a drift of up to 100 ppm is within the design margins of the library (4s in 24h is only about 46 ppm).
I think one problem is the library stays in clock_state 2 (free) even there's now reception for 24 hours.
One possible solution is to add a watchdog: If the input pin doesn't toggle for a specified amount of time it resets the library.

Test three shows the typical behaviour I mean. Board one shows large clock deviation and board 2 wrong date.
This test is not so meaningful as the result changes a bit everytime. For more reliable results more testing is needed.
But it should be quite easy to reproduce by simulating bad dcf77-reception (put the antenna near metal objects or partly wrap it in aluminium foil).

Maybe this way it's possible for you to reproduce the problem !

To finish with something different:
Further up you wrote resetting the library is done by calling DCF77_Clock::setup().
This does NOT work for me, I also have to call DCF77_Local_Clock::setup().

Thanks and best regards,
lakeroe

udoklein, 2018-05-14:
Can you record some trace with the "Swiss Army Debug Helper" in mode Dm ? At least enough to reproduce the issue + some more hours?

With regards to the simulation of bad reception: I tried several things in the past. E.g. the standalone debug helper. So far I found no way to reliably simulate real word noise. I know that in theory a lot of the noise can be modelled, but reality is still different. Without real world data I have no chance. The only thing that I can say for sure that I can not reproduce the issue with my setup. And I tried quite a lot of things (e.g. putting the antenna on top of magnetic loudspeakers or on top of motors). Sometimes I just pulled the antenna for some hours. The issue is that this approach is very time consuming and I can not reproduce the behaviour so far.

lakeroe, 2018-05-14:
Okay, I'll try to post a log to reproduce the behaviour.

Regarding to reset the whole library:
Does it make sense to you to also call DCF77_Local_Clock::setup() ?

udoklein, 2018-05-14:
Good point. Yes, this is a good idea. Maybe I should add it as already suggested in issue #30. Well, maybe if it hits the third time ;)

lakeroe, 2018-05-17:
Just a short intermediate result:
So far I was not able to reproduce the wrong date behaviour using the "Swiss Army Debug Helper".
But I'll keep trying ...

Furtermore there are two more questions:
1) In order to compile your library for an [ARM STM32F103C8T6](https://www.mikrocontroller.net/articles/STM32F103C8T6_STM32_Billig_Board) running at 72MHz  I added one
#if defined(__STM32F1__) #endif section at the end of [dcf77.cpp](https://github.com/udoklein/dcf77/files/2013164/dcf77.txt)
Could you just have a quick look to see if there's any obvious mistake ?
Is there anything to change for the different CPU frequency ?
2) In the meantime did you have a chance to look at my LOG above and do you have an explanation for the big clock deviation ?

lakeroe, 2018-05-17:
Thanks for your analysis.
As I already wrote, I waited for a complete sync and then artifically introduced the signal fading by turning the antenna. My location is near [here](https://www.google.at/maps/@47.9358919,13.5932374).
For me this issue is not relevant at all, because I only need a reliable time once a day and this can be easily achieved by resetting the whole library as mentioned above.

Regarding to double check the local clock 100 ppm design limit. I don't have the possibility to accurate measure the quartz crystal frequency but how about following idea:
I use the 1000 Hz Systick Timer to create a clock and compare it to a reference (www.uhrzeit.org/atomuhr.php). If it deviates less than 8,6s in 24h (=100 ppm) then I assume it's okay ?

nameoftherose, 2018-05-24:
We are just a few kilometers apart ...
What dcf module are you using? How long does it take to sync?
How long it stays in sync per day? (in my case about 8h/d)

udoklein, 2018-05-24:
Just send me your phone number by private mail. Maybe we can have a call during the weekend. With regard to the issue, you say you use systick. Thus I assume you are running on ARM. On Saturday I will setup a test with an ARM board for 24h and see if there are any regressions. With my board I had never any issues so far as the signal quality around here is way above the limit for my library. So either there is some regression that did not catch my attention or there is something with your setup. 100 ppm should be absolutely OK, but there may be other issues. I suggest to discuss this by phone (or in person).

udoklein, 2018-05-26:
So far my tests show nothing unusual. Maybe you could run the debug helper in mode DA to see if your clock is really stable. Sometimes it is not the hardware but interrupts that mess up the timing.

udoklein, 2018-05-27:
After 48 hours on Arduino Due (ARM) with a Pollin Module everything looks perfectly well.

``   168310, +------XXXXXXXXXXXXXXXXXXXX85-+---------+---------+2X9--7---+---------+3--------+---------X---------
confirmed_precision ?? adjustment, deviation, elapsed
0.0625 ppm, @+ , 24.8750 ppm, -1 ticks, 412 min + 38007 ticks mod 60000
57/100~25/600

Decoded time: 18-05-27 7 21:31:43 CEST ..
  18.05.27(7,0)21:32:42 MEZ 0,0 26 p(67457-40:255) s(191-0:26) m(244-5:33) h(241-0:33) wd(249-1:34) D(246-3:34) M(245-1:34) Y(250-4:34) 127,32,32,50
``

I have to assume that it is something with your setup. The question is what?

lakeroe, 2018-05-28:
Sorry for my late reply. I was not at home the last days ...

I could observe the drifting clock on both AVR and ARM based board.
One thing I noticed is to have a clean and stable power supply. When powering the whole setup by a very cheap "usb china power supply" I don't get any reception at all.

As suggested, I will run the debug helper in mode DA and report back.
Furthermore I can offer to send you a complete ARM based affected hardware for your investigations.

I'm quite busy until sunday but next week we can have a call if you like.

@nameoftherose
If you're interested you can send me your exact location by private mail ...

lakeroe, 2018-05-29:
Please find another LOG attached (in DA mode).
Interestingly the date is wrong but the time is accurate.

[dcf_2018-05-28_ARM.zip](https://github.com/udoklein/dcf77/files/2049939/dcf_2018-05-28_ARM.zip)

nameoftherose, 2018-05-30:
This looks very much like with what I experienced. It is not clock drifting, the decoders decode the wrong date.  If computed day and decoded day are not equal the date decoders should be reset and transition to synced state barred. I also experienced hour errors, but very rarely.

nameoftherose, 2018-05-30:
This is a graph of the various performance metrics produced by swiss_army_debug_helper.
[image]
The phase_lock (marked p) is erratic. Assuming the library has been properly ported, I would say it is noise. Filter power supply - I am using LC filters - even use battery or feed from a laptop operating from its battery. Reset the clock whenever calculated week_day defers from the decoded one.
For my location (2000km from the transmitter) a 55mm antenna is too short, but you are much closer.

lakeroe, 2018-05-30:
Thanks for your analysis, you can clearly see the dependancy of the antenna orientation.
At the beginning the phase is locked for some time (antenna is horizontal and pointing towards dcf77 sender).
Then it looses the lock (antenna was turned vertical) and at the end it's locked again (because antenna is horizontal and pointing towards dcf77 sender again).

nameoftherose, 2018-05-30:
I am using the [Conrad module](https://www.conrad.de/de/dcf-empfaenger-modul-641138-passend-fuer-serie-c-control-641138.html). Its antenna is 50mm.

udoklein, 2018-05-31:
Looking at the analysis of nameoftherose there is very interesting behaviour. First sof all the phase lock fluctuates a lot. Then during the period with the poorest phase the month and day decoder "lock" to the noise. Once the signal gets really good (before 17:00) the other decoders increase in quality. However the mistakenly locked decoders start to converge to the proper value. Hence the quality decreases.

The easiest way to fix this mess is to not allow the "slower" decoders to decode before the fast decoders are ready. That is: require that locks can only be acquired in the order phase, seconds, minutes, hours, days, months, years.

I think this would fix it. There is a catch though. The current approach allows for significantly faster startup. I think I will fix it by introducing another configuration flag. What do you think about this proposal?

With regards to using the weekday as some kind of checksum this will unfortunately not fix it. The reason is that if your signal quality is such that this issue happens then introducing this as a checksum will fix it for 6 out of 7 days of the week. The rest there will be still this issue.

There is one more thing that I could do: I could introduce additional "flat out detection bins". The price to pay is 1-2 additional bytes of sram consumption per decoder. Thus about 12 additional bytes. This would also help to relieve this issues.

Thus the action plan would be as follows:

1) Add a flag "optimistic decoder" which toggles the way the decoders are working. TRUE = as currently implemented, FALSE = do not allow the slow decoders to decode before the fast ones are at least marginally ready.

2) If the flag is TRUE use the weekday information for faster transition to "synced", otherwise require improved consistency.

3) Add a flag "flat out detector". If FALSE go ahead as today. If TRUE add additional detection bins for flat out 0 and flat out 1. (but not for minutes or hours, for those only for flat out 1). If the decoder locks to the flat out signal then void all successive decoder stages and do not allow to sync.

What do you think about this approach? In particular about the choice of the default values? Should I default to maximum robustness (paying with more memory and slower initial sync) or should I default to the optimistic setup and require that people with extra poor reception need to adapt the configuration?

udoklein, 2018-05-31:
@nameoftherose By the way: what did you use to plot the log statistics?

udoklein, 2018-05-31:
@lakeroe If you send me your hardware then I also might be able to gain more insights. Did I understand it right that you are within <300 km from Frankfurt?

One more question: why are you putting the antenna intentionally into a poor reception position? This is a very interesting approach but I would assume that the antenna is mounted in an optimal orientation and that the library will only have to tackle "other noise". What you are doing is basically raising the noise floor by more than 10d dB, maybe even 20 dB. This is great for testing but close to Frankfurt it is somewhat pointless. So I am wondering why you do this?

BTW: together with the analysis of nameoftherose I think this finally shed some light on why some people reported issues with my library that I was not able to reproduce. Obviously your test approach is a good idea :)

udoklein, 2018-05-31:
I analyzed the log and with the hints of the picture by nameoftherose the plot thinkens. Your decoder module is biased to all 1. Thus during periods of bad reception it shows a different behaviour than mine. Mine biases to 0. This also explains why I was never able to figure this out. My module is biased differently. I will fix the library. However I am very busy right now. I can not promise that I will be finished in June. This might take some weeks.

nameoftherose, 2018-05-31:
The graph is created as follows:
1. minicom creates a timestamped log
2. the log is proccessed with awk -f [qf.awk](https://github.com/nameoftherose/dcf77-leonardo_support/blob/master-1/logfiles/qf.awk) to produce qf.csv
3. qf.csv is opened in gnumeric
4. [qf.gnumeric](https://github.com/nameoftherose/dcf77-leonardo_support/blob/master-1/logfiles/qf.gnumeric) is then opened, it contains links to qf.csv, and the graph is shown
5. the dates in cells A1 A2 are the graph limits.
This could be done in any spreadsheet.

nameoftherose, 2018-05-31:
Yes I think your action plan is very promising.
Take your time.
Thank you.

lakeroe, 2018-06-04:
@udoklein
Regarding your questions

*) I think your plan sounds promising and we should give it a try.
Once the new software is available for sure I can run further tests and post logs.

*) Default values
I would go for maximum robustness because I think it's more important to have a reliable time than to save some bytes on RAM and faster sync.

*) My hardware
Due to the newest findings do you think it's still necessary to send my hardware to you ?
If yes, another possibility is to by your own hardware. It's just so cheap and probably the same price than sending mine to you and back. The hardware I use is
[Pollin modul](https://www.pollin.at/p/dcf-empfangsmodul-dcf1-810054) 5,45€ + 4,95€ (shipping)
[STM32F103C8 blue pill board](https://de.aliexpress.com/item/STM32F103C8T6-Micro-USB-controller-STM32-Development-ARM-Learning-Board/32832747542.html) 1,85€ from china
Is that okay for you ?

*) Distance from Frankfurt
According to Google Maps I'm 407km away (in Austria near Salzburg).

*) Antenna orientation
I put the antenna in a poor reception position to get a quicker result.
If it's in an optimal position I might never reproduce the drifting time problem ...

*) Time schedule
No hurry and take your time. It's finished when it's finished !

udoklein, 2018-06-04:
Well, I used a Pollin module for my tests --> in my location it is differently biased. As you say most probably I will gain no insights from your hardware. However you could do me a favour. Now that we know what is most likely the issue it would be nice if we can create a log file which will reproduce the issue with my standalone debug helper. Unfortunately the log file with the issue does not. The reason is that the log does not capture the full information prior to the first sync.

There would be two options:
1) I synthesize this information
2) Hopefully easier: you create another log just the way you did when you reproduced the error. But this time instead of turning the antenna after the first sync give it another 20-30 minutes to acquire a better lock, then turn the antenna and wait till the issue reoccurs.

According to the theory (2) should still reproduce the error but it should also capture enough signal such that the standalone debug helper will also recreate the issue. This would help me a lot during testing.

Would this be possible for you? I am not in a hurry. If this takes 2 or 3 weeks it would still be in time for testing.

lakeroe, 2018-06-05:
Sure, I will create another log and report back ...

lakeroe, 2018-06-06:
Please find another LOG attached (in Dm mode).
This time both date and time was wrong ...
Maybe @nameoftherose is kind enough to generate another nice looking graph ...

@udoklein
Could you just have a quick look if it fits your need ?

[dcf_2018-06-05_ARM.zip](https://github.com/udoklein/dcf77/files/2077726/dcf_2018-06-05_ARM.zip)

nameoftherose, 2018-06-06:
[image]
I  would not say the clock is drifting, the algorithm decodes the wrong time (line 638370) and syncs to it. Synchronization is then lost and reacquired, this time correctly (line 658922).
This log confirms @udoklein analysis.

udoklein, 2018-06-07:
@lakeroe - Excellent. The file is does reproduce the issue with the standalone debug helper. As it now turns out even the first one did but I mistakenly used a wrong parameter setup. Sorry for causing duplicate work. However on the upside I have now more test data :)

With regards to the plots I added a log_2_csv converter in the latest commit here: https://github.com/udoklein/dcf77/blob/master/extras/standalone_debug_helper/log_2_csv.py  
The tool has a command can convert logs to csv files. Then any plot program should be able to consume it. Thanks to @nameoftherose for bringing up this idea. I still feel somewhat humbled that this did not occur to me earlier.

**Attention:** I also changed the log format slightly. Thus it does not work with version of the library older than v3.2.8. In case of logs written with previous versions of the library all occurences of "MEZ" must be replaced by "CET" and all occurences of "MESZ" must be replaced by "CEST" before converting log files.

udoklein, 2018-06-07:
By now I think I will patch 

`static void process_single_tick_data(const DCF77::tick_t tick_data)`

The point is that in

`static void sync_lost_event_handler()`

I wipe the slower decoders once a sync is lost. But then I do not prevent them from gaining "quality" even if the fast ones fail to do so. The point is that this situation is with exceedingly high probability caused by biased noise. If the noise is random, then this can not really happen.

As a consequence I would not need an additional flat out detector at all. The fast stages already provide this information.

I am now pondering what happens if the fast stages have very high quality and the slow stages are not yet there. Now if biased noise enters the clock then I can not detect it and it will again pull the slow stages into this undesired state.

Of course I could add a check for "jumps" in time. But then what to do? Maybe reset the clock.

I am still thinking. The implementation of the fix will be much easier than finding it.

udoklein, 2018-06-07:
On the other hand maybe a flat out detector could leverage the timezone bits + the 1 minute digit bit. Thus every two minutes it should see 3 ones and 3 zeroes. If this is not the case then the signal might be biased. Now I need to figure out how to prevent it from a "random walk". Probably an exponential filter approach would be good. This raises the question for a suitable filter constant.

udoklein, 2018-06-09:
With regards to the flat out detector it might be actually better to not look into any of the decoded bits. Probably it would be best to look into the raw signal. E.g. it might be a good idea to look at the first 100 ms and the 500-600 ms interval. The pattern might be

0 0 --> flat 0
1 1 --> flat 1
0 1 --> drowned by noise
1 0 --> everything the way it should be

This would have the significant advantage of a very short latency for detecting flat outs. This leaves the question what would be a good fitler constant. --> Need to analyze the data of the examples.

Jo-Achim, 2018-06-12:
Great work here!

udoklein, 2018-06-23:
Release [v3.2.9](https://github.com/udoklein/dcf77/tree/v3.2.9) introduces some slight change in the way the decoder templates are parametrized. This "optimization" is part of the preparation for the fix of the current issue.

udoklein, 2018-06-30:
I am right now testing v3.2.10. Here are some results if I feed the logs lakeroe into it.

[image]
[image]
[image]
[image]
[image]

udoklein, 2018-06-30:
Picture 1 and 2 show the new library for the olders two files. Picture 3 and 4 show library 3.2.8 vs. 3.2.10 vs. the newest log file. As it turns out the library still struggles with this file. Picture 5 shows why. I took only every 60th sample. Thus the second should be stable. However it is not. The library acquired a phase lock but the second is drifting. What ???????

I double checked the original log file. This is not a bug in the library. This is the hardware which is causing the issue. Actually a one point between 86000 and 88000 seconds its drifts by almost 1 second in 500 seconds or 2000 ppm. Way out of spec.

I would like to point out two things:
1) In the last case there is either some hardware flaw with the local oscillator or there is some other code running that messes with the interrupts.
2) Because of the fix the new library takes longer to actually lock to a signal. Thus if there are only very short periods with a reasonable good signal, then it may not lock to the signal at all. There is not a lot that I can do about this. The library is statistical in nature. Everything that I do to reject bad signals will decrease the chances to accept correct but noisy signals. I hope the change is an overall improvement.

Please test and provide feedback.

udoklein, 2018-06-30:
If you are testing either use [v3.2.10](https://github.com/udoklein/dcf77/releases/tag/v3.2.10) or, if you use v3.2.11 make sure that you set https://github.com/udoklein/dcf77/blob/v3.2.11/dcf77.h#L110
to the value used in v3.2.10. E.g.

`enum controller_minute_quality_threshold_t : uint8_t { aggressive_minute_quality = 0, standard_minute_quality = 2, conservative_minute_quality = 4, paranoid_minute_quality = 6 };
    static const uint8_t unacceptable_minute_decoder_quality = controller_minute_quality_threshold_t::conservative_minute_quality;
`

brettoliver, 2018-06-30:
Hi Udo.
Thanks for the update. 
I am in the south of England. This time of year once the temperature rises I get very bad signal reception in the afternoon into the evening.  
I have 9 DCF77 clocks running off 1 Aerial via a Digital repeater so they all get the same signal. One clock runs library version 2 and never has any issues the other clocks run the old version 3 library. One of the clocks has a problem when the signal gets bad over a long period and losses Sync showing Lock on the display. Eventually the time and date go way out and the clock never returns to sync even with a good signal.

I have now loaded this clock with you new library 3.2.10 and will see how it goes. 
The DCF77 Signal is very bad at the moment so I expect a long wait for sync but I don't mind this if it fixes the Sync issues.

I'll let you know how I get on.

Regards.
Brett.

udoklein, 2018-06-30:
Hi Brett,
if you could record one or two days with "Swiss Army Debug Helper" in mode Dm, then I could analyze what happens. This is the only way for me to get real world noise. As it becomes more and more obvious my simulator does not simulate the real world situation. Hence I require these logs to improve the library further.
Regards, Udo
P.S. Which exact version 2 are you using? Maybe I can spot the difference in the code and port the different behaviour to version 3 again.

brettoliver, 2018-06-30:
Hi Udo.
I will be moving the clock that has sync issues in the next couple of weeks. I will set it up near my PC and record the data for you.
The version 2 code I use on my Master Clock is v2.0.4. I think this has been running for over 4 years now with no errors.
Regards.
Brett

brettoliver, 2018-06-30:
Hi Udo.
My clock has now synchronised after 8 hours of intermittent bad signal. See my signal here 1 = 100% 0 is <100% http://home.btconnect.com/brettoliver1/Geiger_Counter/Charts.htm

The only problem is the clock has synchronised to the wrong time. The seconds are correct but the time shows  06:09:45 20/04/2000 not 20:09:45 30/06/2018.

I would normally run an EEPROM erase program at this stage to get the clock to sync correctly. 
 http://home.btconnect.com/brettoliver1/Arduino_EEPROM_Erase.htm

What do you think?
Brett.

brettoliver, 2018-06-30:
Have run EEPROM Erase as clock went to Lock. Clock has now  syn'd correctly after only a few minutes.

udoklein, 2018-07-01:
Hi Brett,
I have no idea why erasing the EEPROM would help. Maybe this is just coincidence, maybe not. I definitely need log files to gain more insight. I pushed a new version of the library (v3.2.12) with slightly enhanced logging. If you could use this and create some log files maybe I can gain more insights.
Regards,
Udo

brettoliver, 2018-07-01:
Hi Udo.
I nearly always have to erase the the EEPROM once any of my clocks have lost sync. I presumed the learned "settings" were stored in EEPROM and got reloaded when the Arduino was restarted. Once these "settings" are incorrect I find my clocks won't sync to the correct time even after reset.
Once I do an EEPROM erase the clocks always re-sync correctly.

I will be able to use your "Swiss Army Debug Helper" now as I have found a long USB cable to reach my clock (it's a Longcase Clock so I can't move it easily).

I have loaded the  "Swiss Army Debug Helper" into my clock and sent "Dm" to the clock.
I am showing a "scope" type output on the serial monitor.

How do I save a log file?
Brett.

lakeroe, 2018-07-02:
Hi Udo,

I tried v3.2.11 (with default parameters) and ran my "usual" test (put the antenna in an optimal position, wait for the best clock_state possible and then put the antenna in a non-optimal position). After 33229 seconds the library decodes the wrong time ...

Could this be due to local clock drift you assumed above ?

Best Regards,
lakeroe

LOG-File: [dcf_2018-07-02_ARM.zip](https://github.com/udoklein/dcf77/files/2154009/dcf_2018-07-02_ARM.zip)

lakeroe, 2018-07-02:
In my first understanding this setting only effects the initial lock time, so I've left it at the default value.
But now I think you're right.
So I change this setting and do some more testing ...

Jo-Achim, 2018-07-04:
Hello Udo,
hello together,

I would like to get involved in this topic, although I did not understand all the details of the problem / solution.

In my understanding, the current problem solution (v3.2.1x) has, compared to the 'normal'-decoder-mode, the problem of a very long initial decoding of the DCF signal, which is, why the maintenance of the 'normal' configuration of the decoder (Out-of-the Box) makes sense.
Only in especially bad reception situations with corresponding noise is a configuration change of the decoder to the new configuration (v.3.2.1x) useful. Correct?

So my question: Can the decoder be changed during operation to the new configuration; e.g. about changing a variable?
If so, would be the following solution possible? ...
The decoder starts in 'normal' mode and stays in it (the first decoding always seems to be correct!?).
It is up to the user whether and how he integrates a kind of time and date plausibility check into his sketch. For a wrong year, it is probably the easiest; everything that is not a leap from year + 1 and on 31.12., is wrong - and leads to switching the decoder in the 'safest mode', i.e. the current solution (v 3.2.12).

The next question is, whether there must be a switch back to the 'normal' mode of the Decoder? Which raises the question of whether a solution might also consist in the fact, that the decoder after first decoding switches in the 'safest mode' and remains there?
If this is not ideal or desired, the Date / Time plausibility test must downshift in the 'normal' mode within appropriate intervals.

If it should be possible that the decoder could independently determine a signal quality that leads to the date / time error - which I do not expect because of the effort - then the decoder could ideally perform the toggling itself.

Maybe there will be more approaches...

udoklein, 2018-07-04:
@Jo-Achim as you stated yourself you lack understanding of the issue. We have no lack of questions or ideas. If you want to contribute, please tell me:

- What is your background? 
  - Do you have C++ coding skills? Which level?
  - Do you have signal processing, statistical or mathmatical skills? Which level?
  - What else do you think you could contribute? 
- How much effort would you be willing to contribute?
- What is your stake in this project? Why do you want to contribute?

lakeroe, 2018-07-05:
I did some more testing by setting the unacceptable_minute_decoder_quality variable to different values.
Furthermore I used 3 different antenna positions:

1. Best position  - antenna horizontal and rotated by 90 degrees towards Frankfurt
2. Worse position - antenna vertical
3. Worst position - antenna horizontal and pointing directly towards Frankfurt

Results

1) aggressive_minute_quality
Time for initial lock (correctly decoded date/time) in antenna position 1: 546 s
Antenna position 2: Wrong date/time after 33229 s (old behaviour)

LOG-File: [dcf_2018-07-01_ARM.zip](https://github.com/udoklein/dcf77/files/2165264/dcf_2018-07-01_ARM.zip)

2) standard_minute_quality
Time for initial lock (correctly decoded date/time) in antenna position 1: 986 s
Antenna position 2: No time drift
Antenna position 3: time starts to drift after a few minutes

LOG-File: [dcf_2018-07-02_ARM.zip](https://github.com/udoklein/dcf77/files/2165265/dcf_2018-07-02_ARM.zip)

3) conservative_minute_quality
Time for initial lock (correctly decoded date/time) in antenna position 1: 1618 s
Antenna position 3: time starts to drift away (off by about 130 s at the end of the log)

LOG-File: [dcf_2018-07-03_ARM.zip](https://github.com/udoklein/dcf77/files/2165266/dcf_2018-07-03_ARM.zip)

4) paranoid_minute_quality
Time for initial lock (correctly decoded date/time) in antenna position 1: 1720 s
Antenna position 3: time starts to drift away (off by about 130 s at the end of the log)

LOG-File: [dcf_2018-07-04_ARM.zip](https://github.com/udoklein/dcf77/files/2165263/dcf_2018-07-04_ARM.zip)

So in my opinion you can clearly see an improvement although the time drift is still present.
But I still miss one feature:
Resetting the whole library (see the title of this thread) is still not implemented ;-(

However a big thanks for your great work here.

nameoftherose, 2018-07-05:
@lakeroe , is position 3 a new position or have you used it in the past?
What antenna position was used in the previous logs?
In log no 2 how long have you kept the antenna in position 2?
A drift of 130s in ~8.5hours corresponds to 4300ppm. This is way too much. Are you sure your PC remained synchronized during the measurement period?

lakeroe, 2018-07-05:
> Is position 3 a new position or have you used it in the past?

Position 3 is not new, I've also used it in the past sometimes.
However during those tests I found this is even worse than putting the antenna in an upright position.

> What antenna position was used in the previous logs?

I used position 2 (at least I think so).

> In log no 2 how long have you kept the antenna in position 2?

I think about 1h in position 1, 23h in position 2 and the rest in position 3
In logs no 3 and 4 I skipped position 2 completely.

> A drift of 130s in ~8.5hours corresponds to 4300ppm. This is way too much. Are you sure your PC remained synchronized during the measurement period?

Yes I'm sure, my reference time is this: https://www.uhrzeit.org/atomuhr.php

nameoftherose, 2018-07-05:
@lakeroe 

>However during those tests I found this is even worse than putting the antenna in an upright  position.

 It seems to me you are changing the rules. For position 2, which was the standard so far, this is a huge improvement. Have you encountered in practice a situation worse than position 2?
Bear in mind that due to the statistical nature of the problem no algorithm will be perfect. We can only lower the probability of error.
The behaviour in logs 3,4 is very different from what I observe, therefore I cannot comment.

brettoliver, 2018-07-06:
Sorry everyone, I seem to have lost some notifications on this thread and missed some replies.

Udo said 3 days ago.
"@brettoliver: the behaviour you are experiencing might be a combination of two issues:

You local oscillator is somewhat unstable. In particular it might have a significantly different frequency after a reset vs. long runtine.
Due to the autotuning with the large tolerance it tunes to the frequency that settles after a long runtime. Then after a reset it will try to immediately lock to the persisted frequency. This is of course impossible if the oscillator drifted a lot.
If my theory is right then this can be remedied as follows:
In dcf77.h change
static const bool has_stable_ambient_temperature = true;
to
static const bool has_stable_ambient_temperature = false;
and tell me if this fixes the issue. If not so I will change the library such that for "false" it will not persist the tuned frequency anymore. This is the "big hammer" option that I would like to avoid but this effectively resets the EEPROM on each new start."

Udo.
I have reconnected my Longcase Clock DCF decoder board and loaded the code back with your modification to the dcf77.h file.

I will run it over the next few days and see how it goes.

Your theory sounds good. I always wondered why some of my nine DCF connected clocks although connected to an identical signal (I use a digital repeater off 1 common DCF77 receiver) would fail and not re-sync after a long period of bad signal.

I like the "big hammer" option as a reset will always bring a clock back to life without having to run the EEPROM clear program. This is especially true if many years down the line you have an old clock running outdated library versions fails and you have no compatible hardware/software to perform an EEPROM clear and reload the Arduino.

udoklein, 2018-07-06:
@brettoliver [v3.3.0](https://github.com/udoklein/dcf77/tree/v3.3.0) introduces the biggest possible hammer for your issue. I completely removed the EEPROM persistency.

@lakeroe thanks for the logs. I may need some time till I can analyze them. I will be somewhat busy the next several weeks. @nameoftherose is most probably right though. And if your oscillator drifts significantly more than 100 ppm, then you are using the library way out of spec. As a consequence there are no guarantees whatsoever. I would expect that it only decoded garbage. @nameoftherose is also right about the statistical nature of the library. There is no simple means for me to sort out the proper signal if the quality is poor. If the target platform would be a bigger system with lots of CPU and memory and not some poor microcontroller with 2k SRAM, then this would be much easier. However the whole idea was to provide a very noise tolerant library in spite of such constrained resources. 

If the logs indicate that my fix was not really good, I will try to improve it. But I doubt that I can much more. 

Conclusion: optimize your antenna position.

nameoftherose, 2018-07-07:
@lakeroe , Have you observed drifting with ATMEGA? Will you consider creating a timestamped log? (but only if you are **absolutely sure** the logging computer maintains **correct** time)

udoklein, 2018-07-07:
@lakeroe @nameoftherose In my opinion the drift is clearly visible in the logs. No need for a timestamped log. The point is that the scope view shows that it drifts. As long as the library maintains its phase lock it is clear that the signal is "timestamped" by DCF77. Thus by definition (because DCF77 transmits the legal time as maintained by the PTB)  any shift of the phase is a deviation from nominal frequency.

udoklein, 2018-07-07:
Well, this is the point. The library was ported by me (and tested) for Arduino Due. I had it under test several weeks. So I assume it should be OK. However lakeroe seems to be using a different board. On the other side he also tested with AVR.

Once I find a little bit more time I will analyze the new logs and see what I can find.

Jo-Achim, 2018-07-12:
One, despite my ignorance ;-), maybe helpful random observation as information:

Unknown, therefore hardly repeatable and so first occurred circumstances, led to the fact that two quartz MiniPro (each ~ 30ppm) with the same software but different DCF77-Libraries different times and a different date indicated. Both MiniPros are powered by the same DCF77 receiver!

**MiniPro_1 with Library <= v. 3.2.8 and >= 3.2.2** (out-of-the-box), unchanged for some time (therefore rather < v. 3.2.8), shows - verified - the correct time (and date).
**MiniPro_2 with Library v. 3.3.0** (out-of-the-box) initially shows only an incorrect time, later also an incorrect date; EEPROM explicitly deleted.

What happened? An inadequate, very noisy DCF77 signal resulted in 'unlocked' reception conditions within about 24 hours.

MiniPro_1 was initially affected; Synced -> Locked -> Unlocked. MiniPro_2 followed clearly delayed; Synced -> Locked -> Unlocked. Worst decoder mode for both: Unlocked.
The time interval with the MiniPro_2 followed is unknown (the decoder modes are signaled by colored LEDs, time or date were not checked at first). But it is from 10 minutes and (much) more likely to go out; it could also have been 1 - 2 or more hours.

After returning to perfect reception conditions, MiniPro_1 was the first to return to sync mode; an incorrect time or date was not observed.
While MiniPro_1 was back in sync mode, MiniPro_2 was still in unlocked mode. 

In the individual observations consistently good reception conditions, the following data emerged (the date of both MiniPro was on Wed., 11.07.2018 identical correct):

**Wed., 11.07.2018:**
**MiniPro_1 = 12:43:10 (synced);	MiniPro_2 = 11:58:36, Wed., 11.07.18 (unlocked).** 
MiniPro_1 = 13:00:00 (synced); 	MiniPro_2 = 12:15:26, Wed., 11.07.18 (unlocked).
  ...
MiniPro_1 = 13:45:00 (synced); 	MiniPro_2 = 13:00:26, Wed., 11.07.18 (unlocked).

MiniPro_1 = 13:51:00 (synced); 	**MiniPro_2 = 13:06:26, Wed., 11.07.18 (locked).**
MiniPro_1 = 14:00:00 (synced); 	MiniPro_2 = 13:15:26, Wed., 11.07.18 (locked). 
  ...
MiniPro_1 = 21:00:00 (synced); 	MiniPro_2 = 20:15:26, Wed., 11.07.18 (locked).

<no further observations on 11.07.2018>

**Thu., 12.07.2018** (time and date of MiniPro_1 still correct):
MiniPro_1 = 03:29:04 (synced); 	**MiniPro_2 = 03:04:05, Tue., 18.07.79 (locked).** 
MiniPro_1 = 08:02:07 (synced); 	MiniPro_2 = 07:37:08, Tue., 18.07.79 (locked). 
  ... 
MiniPro_1 = 14:00:00 (synced); 	MiniPro_2 = 13:35:01, Tue., 18.07.79 (locked). 
  ...

The above last state continues to persist; including time and date offset. I will supplement if MiniPro_2 has reached 'synced' mode to estimate the total time until then.

Notes: The synchronization bit is error-free "0" (without visible interference). Initial initialisation takes about 6 - 7 minutes with approximately the same good reception conditions.

Conclusion: According to my impression, the Library v. 3.3.0 (compared to previous versions) buys the longer stability with deteriorating signal with a very long time back into sync mode; with - in this case - wrong time and date.

And sorry, no logs.

udoklein, 2018-07-12:
@Jo-Achim Your "analysis" is not helpful. Either you provide logs for analysis or you shut up. I asked some pretty simple questions about your capabilities that you did not answer so far. Please provide some answers to my questions or provide reasonable input for analysis. Please do not provide insights like "the improved version of the library is somewhat better but not good enough". This does not lead to any insights and does not help with any analysis at all. If you continue "contribute" the way you do right now I will remove you from this conversation.

[31 more replies]

closed 2018-08-11 by udoklein