udoklein/dcf77 #32: Feldstärke und monitor_pin

MrGlasspoole, 2018-05-24:
Hallo,

was genau sagt monitor_pin aus?
Heißt das nur das irgendwas vom DCF77 Modul kommt oder bedeutet es das das Signal wird empfangen?

Die Meinberg Empfänger haben eine LED für Feldstärke.
Desto heller sie leuchtet, desto höher ist die Feldstärke.
Ist das vergleichbar mit prediction_match?

udoklein, 2018-05-24:
monitor_pin ist keine Eigenschaft der Library sondern in einigen Beispielen gibt es eine "monitor_led". Eine Codesuche bringt Dich direkt zu solchen Zeilen: [https://github.com/udoklein/dcf77/blob/1103b38718e8be451031f6b9986a4291c0a085e0/examples/Swiss_Army_Debug_Helper/Swiss_Army_Debug_Helper.ino#L122](url)  Da siehst Du, daß das lediglich den Input anzeigt "monitored".

closed 2018-05-24 by udoklein

MrGlasspoole, 2018-05-25:
Und Feldstärke?

udoklein, 2018-05-25:
http://www.catb.org/esr/faqs/smart-questions.html

MrGlasspoole, 2018-05-26:
Für solche Antworten hab ich das Buch gekauft...

udoklein, 2018-05-26:
1) Mein Buch behandelt einfache Lichteffekte.
2) Die DCF77 Library hat mit dem Buch nichts zu tun.
3) Du sagst nicht genau was Du tust, was Dein Hardwaresetup ist und schreibst nicht einmal vollständige Sätze als Frage. Ich habe keine Glaskugel mit der ich raten kann was genau Dein Problem ist.
4) Du hast anscheinend die Antwort nicht gelesen / verstanden. Wenn es Dir nicht wert ist wenigstens eine vernünftige ausformulierte Frage zu stellen, wieso sollte ich mir die Mühe machen zu raten was Du vieleicht willst?
5) Die DCF77 Library würde nach den üblichen Stundensätzen als Auftragsarbeit deutlich über 50 000  Euro kosten. Du lädst Dir also kostenlos eine ziemlich hochwerte Library runter, und erwartest, daß Du kostenlosen Support bekommst weil Du mein Buch (zu einem anderen Thema) gekauft hast. Boah, Du glaubst anscheinend damit wird man reich. Pro verkauftem Exemplar habe ich weniger als 5 Euro in meiner Kasse.
6) Du weisst indirekt darauf hin, daß Du für Geld Unterstützung erwartest. Dir fehlt gutes Karma: https://blog.blinkenlight.net/good-karma/

MrGlasspoole, 2018-06-02:
Was es an

> Die Meinberg Empfänger haben eine LED für Feldstärke.
Desto heller sie leuchtet, desto höher ist die Feldstärke.
Ist das vergleichbar mit prediction_match?

anders zu Formulieren gibt weiß ich nicht.

Das Buch hab ich gekauft um den Entwickler (Dich) zu unterstützen und um eben nicht Fragen zu stellen ohne etwas dafür zu geben. Das mit PayPal habe ich vorher nicht gesehen.

Aber ist jetzt auch egal.

udoklein, 2018-06-02:
Deine letzte Frage war "Und Feldstärke?".

Das Problem ist nicht nur die Formulierung sondern die Anspruchshaltung. Du kaufst ein Produkt beim Franzisverlag und _verlangst_ dann Unterstützung bei einem Open Source Projekt ohne Deine Hausaufgaben zu machen.

Lies Dir nochmal genau die FAQ von Eric durch und stelle eine brauchbare Frage. Was die Feldstärke angeht: versuch mal mit Deiner Hardware einen Feldstärkeindikator zu bauen. Versuch mal zu verstehen warum Du das nicht hinbekommst und wie Du zu der Auffassung kommst, daß das in meinen Beispielen drin sein könnte oder sollte. Danach weisst Du was mit Deiner Frage das Problem ist.