udoklein/dcf77 #37: Compiles, uploads but doesn't run on black pill

ratkins, 2019-03-01:
I'm attempting to run the library on an STM32 "[black pill](https://wiki.stm32duino.com/index.php?title=Black_Pill)". I see #defines in the code, and a couple of closed issues here, indicating at least one other person has made this work. There are two issues:

1) The Arduino IDE places the dcf77_xtal examples under an INCOMPATIBLE submenu, indicating the "Generic STM32F103C series" board's architecture isn't supported. I think fixing this is just a matter of adding "arm" to the `architectures` list in `library.properties`?

2) In any case when I select one of the examples (like DCF77_Scope) and change the pin assignments for my board, it compiles and uploads fine but doesn't run. I don't get any output in the serial monitor and the LED doesn't flash. I've made both of those things work independently with other sketches uploaded by the same method so I'm pretty sure the board itself works fine.

Any idea what I could try to debug this?

udoklein, 2019-03-04:
I was reluctant to add STM32 support because I have no boards to test it. Now you confirm that there are issues. My stance is: if you figure it out and say it is good for you I would merge a pull request. But I am sorry I have no idea how to debug this.

I also do not know what happens if I add arm to the architectures. Since you state that it does not work 100% I would rather not add it. Once I add it people will rightfully expect that it works.

ratkins, 2019-03-04:
I've got a bunch of STM32 boards here and would be happy to send you one if you're interested (no obligation to actually fix my issue, of course.)

udoklein, 2019-03-05:
Yeah, you can send me a board. However at this time I am moving to a new flat --> I will be slow to respond. Also I will not send the board back when I am done. If this is OK for you then send me a board. Is the board supported out of the box by the IDE or do I need something special to install?

ratkins, 2019-03-06:
You can either [upload directly](http://wiki.stm32duino.com/index.php?title=Uploading_a_sketch) with a USB serial adapter or [flash a bootloader](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Flashing-Bootloader-for-BluePill-Boards). I've made it work with both methods, but it's a bit fiddly (esp. on macOS, there's no one place with all the instructions in a sensible order.)

Send me an email (my GitHub handle @fastmail.fm) with your address.

udoklein, 2019-03-10:
My Mail account is currently inaccessible. You can find my address here though: https://blog.blinkenlight.net/licenses/

udoklein, 2019-06-21:
As I said I will take very long because I have some issues :( Anyway today I found some time to look into it. The latest merge should have added support for blue pill which should be compatible with black pill. Do you have version 3.3.3? If so, does the DCF77 scope work as it should? What output does it generate?

I am still digging in my huge pile of electronics for some 3V3 serial adapter. I know I own at least 3, but I am still digging. So I have not yet tested on my own. Anyway can you tell me which version of the library you are using?

nameoftherose, 2020-02-27:
You might like to know that the library compiles for and runs on the Arduino Nano 33 IOT.  Incidentally 33 IOT is crystal-less.
No modifications to the DUE code were needed, one simply has to also enable the relevant code in dcf77.cpp for the `__SAMD21G18A__ ` mcu.

udoklein, 2020-02-27:
Interesting. Does it lock to DCF77 without a crystal? Or does it just execute?

nameoftherose, 2020-02-28:
It **does lock** to DCF77. On closer investigation SAMD21 may derive a reference clock from USB and Arduino seems to use this feature. This means that if it is used standalone, clock stability will suffer. During the test Nano 33iot was connected to a PC.