udoklein/dcf77 #39: Partly initialized structure in dcf77.h

nameoftherose, 2019-07-31:
In dcf77.h lines 2232 -2239 you define structure `DCF77_Encoder now;` and partly initialize it, then in line 2240 you call `now.debug();` which also uses the uninitialized members `uses_summertime`, `leap_second_scheduled` and `timezone_change_scheduled` producing misleading output. I think the following three lines should be inserted after line 2239:

    now.uses_summertime=Flag_Decoder.get_uses_summertime();
    now.leap_second_scheduled=Flag_Decoder.get_leap_second_scheduled();
    now.timezone_change_scheduled=Flag_Decoder.get_timezone_change_scheduled();

udoklein, 2019-08-02:
Thanks for reporting this bug. I will fix it as soon as I find some time. Although easy to fix this may take up to 6 weeks because I am right now extra busy with other stuff. But definitely thanks a lot for finding this detail.

udoklein, 2021-02-28:
This was fixed in 2019 by https://github.com/udoklein/dcf77/commit/a40551ba7a99032db0ff5d8117d5279da0e7bab1 but I forgot to close the issue. I will close it now.

closed 2021-02-28 by udoklein